NHMDenmark/DaSSCo-Digitization #6: Herb03 Imaging Station Computer Issues - collecting experiences for a KU-IT ticket

CelinePLB, 2025-05-28:
I have created this log to collect reports of any issues experienced with the computer at the imaging station Herb03. We plan to submit a ticket to KU-IT to address and hopefully resolve some of the recent problems. Please provide as many details as possible when describing your experience.

On May 27th, while a TIFF conversion of 484 images was running in the background, the computer became extremely slow. It took 3+ minutes for each image to appear in Capture One, and a few images were skipped even though I heard the camera shutter sound. I imaged 60 sheets and then left the computer to finish loading the images while I worked on other tasks. Later, I verified that all barcodes were correctly imaged and saved and imaged a few sheets that had been skipped again. Additionally, the computer was very slow when opening folders during this time. The TIFF conversion took 33 minutes. The computer was working slowly through the entire imaging session. However, while running an ingestion client and imaging earlier the same day, I had no issues with the computer being slow.

annejodobel, 2025-06-17:
On Friday the 13th the following issues was noted:

- The camera was auto focusing while using live view. 
- The camera was non-responsive 4 times in 2 hours and had to be restarted every time.
- The camera shut down while in use (Temp. recorded as 33 degrees)
- White balance couldn't be applied to all photos (10 photos locked). After having located the files, moved them to a new folder they could be rotated and white balance could be applied.
- Capture One showing a notice that 'Capture One path not found'. I am not sure what this means.

Anne

CelinePLB, 2025-06-26:
I had the following issues on Tuesday the 24th of June:

- The camera shut down while in use 3 times (temp. recorded as 33-34 degrees). 
- Many images had to be rotated manually (I estimate that this happened ~10 times more often than usual)
- 10 photos (out of 990) were locked and white balance couldn't be applied. After having located the files via dRAW and moved them to a new folder they could be rotated and white balance could be applied.

CelinePLB, 2025-07-04:
On May 27th, while a TIFF conversion of 484 images was running in the background (33 min), the computer became extremely slow and continued to work slowly afterwards:

- It took 3+ minutes for each image to appear in Capture One, and a few images were skipped even though I heard the camera shutter sound.
- The computer was very slow when opening folders. 

While running an ingestion client and imaging earlier the same day, I had no issues with the computer being slow.